embench/embench-iot #1: configure failed whether or not add --with-chip

quxianmiao, 2019-06-11:
When run ./configure according to INSTALL, the output is as follows:
checking build system type... x86_64-pc-mingw64
checking host system type... x86_64-pc-mingw64
checking target system type... x86_64-pc-mingw64
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... no
checking whether make supports nested variables... no
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether make supports nested variables... (cached) no
configure: error: Chip config directory "generic" does not exist

Then I check the code in configure, No. 3142 wrote if we can't find $srcdir/config/$arch/chips/$chip, the error will be reported.
So I add the option --with-chip=size-test-gcc, it still reported an error:
configure: error: Chip config directory "size-test-gcc" does not exist

After that, I tried --target and --host options. It still did not work well. So I wonder if there is a problem with the script, or is it my usage is wrong?

jeremybennett, 2019-06-11:
Hi Cooper,

I realize I must modify the INSTALL file - I forgot to fix it. Please try
the instructions in doc/README.adoc.

HTH,

Jeremy

jeremybennett, 2019-06-11:
Also, make sure you toolchain is on you PATH.

quxianmiao, 2019-06-11:
I have run successfully, then how should I view the size-test results?

quxianmiao, 2019-06-11:
Thank you for your reply, I have run it successfully .

jeremybennett, 2019-06-11:
Excellent. You should find the results as a table in
<builddir>/testsuite/beebs.log. there should be a table for size results
and one with speed results.

Best wishes,

Jeremy

jeremybennett, 2019-06-14:
I think you have everything working, so I'll close this issue. Don't hesitate to open a new issue if you run into other issues.

closed 2019-06-14 by jeremybennett