embench/embench-iot #3: commercial friendly license

lbmeng, 2019-06-14:
I see currently Embench is licensed under GPLv3.

For Embench to be widely adopted, can we consider switching to a commercial friendly license like BSD or MIT?

jeremybennett, 2019-07-02:
@lbmeng Hi Bin Meng,

I'm sorry this will not be possible. Embench is built from many other benchmark programs from many other authors and teams of authors. GPL is the only possible license we could use, compatible with all the other licenses.

However, I'm not sure this is not commercially friendly. This is an open benchmark suite, so would always be freely available. Is there a scenario where you can see this being a problem?

Best wishes,

Jeremy

lbmeng, 2019-07-13:
@jeremybennett Thanks for the reply. I am evaluating the possibility of porting this benchmark suite to a commercial closed source OS hence the license question.

jeremybennett, 2019-07-15:
@lbmeng 
IANAL, but that shouldn't be a problem. Even open source OS don't proscribe what you can run on them, which is why there is so much proprietary software that runs under Linux.

I'm going to mark this issue as closed, but please reopen if you want to continue the discussion.

closed 2019-07-15 by jeremybennett