embench/embench-iot #6: Run benchmarks on custom system

paul70078, 2019-08-01:
I'd like to run the benchmarks on a softcore cpu on a fpga. So I need to add a custo startup routine and run custom scripts to run it on the processor. Is this possible with the existing systems? Do I manually need to link the object files of the benchmarks with my link script and the startup file to generate the executable or are there better ways to run a custom benchmark?

jeremybennett, 2019-08-05:
This should be possible, but probably harder than it needs to be. As noted in Issue #7,  I'm working on Python replacements for the current build/run environment, which should make this all work smoothly.  Feedback very welcome.

jeremybennett, 2020-04-30:
@pagdot Does the new Python infrastructure make this easier for you?

sobuch, 2020-07-29:
Hello,
I am now able to run the benchmarks on SweRVolf (SweRV EH1) on nexys A7 FPGA. I am using OpenOCD to load programs into the memory and then reading the results from UART. Results therefore need to be measured and printed in functions from boardsupport.c. Besides that I created new run module with the three functions (get_target_args, build_benchmark_cmd, decode_results) defined. Only difficulty was that the current interface used by run modules is too constricting. In this use case it is not enough to run a single command. It would be more convenient for me to define some function to run benchmark directly instead of just passing a command to do so from the build_benchmark_cmd function.

PaoloS02, 2024-03-04:
I'm closing this has it has been inactive for a while now.

closed 2024-03-04 by PaoloS02