embench/embench-iot #7: Specify target architecture

paul70078, 2019-08-01:
I've tried to compile a riscv32 config, but when using `--host=riscv64-unknown-elf`for the configure script it searches for a risc64 config. With the current RISC-V GCC toolchain it seems more common to specify a 32bit architecture with `-mabi` and `-march` instead of compiling a separate 32bit toolchain

Is it possible to compile a riscv32 config without putting it in a riscv64 folder?
I've tried using the `--target` parameter for `configure`, but it still failed to find my config.

jeremybennett, 2019-08-05:
Hi @paul70078 
Thanks for the comment. It's true that GCC is now a single compiler, and you configure with one or other as the default. Most people still stick to a riscv32-*-* or riscv64-*-* naming convention to indicate the default.

There should however be nothing stopping you using `riscv32` as the architecture (so you pick up the architecture, chip and board specifications from the correct directory) and then specifying a different C compiler on the `make` command line `make CC=...`.

This should soon become obsolete. We don't need the complexity of autotools for compiling a benchmark suite, so it will be replaced by a python script in the near future. Similarly we don't need the complexity of DejaGnu to run the tests (which restricts us to Linux only) and that will be replaced by python scripts.

I have work in progress in my own fork of this directory. Have a look at https://github.com/jeremybennett/embench-iot and the `py-build` branch.  The scripts currently named `build-all.py` and `benchmark-size.py` will respectively build the benchmarks and report on the size statistic.  Still under development (notably documentation), but `./build-all.py --help` will tell you the options and you should be able to specify separate architectures, compilers and flags for both compiling and linking.  Once I'm satisfied it's working well, I'll merge into this repo.

jeremybennett, 2019-11-14:
@pagdot I hope the improved documentation helps you. I am going to mark this closed.

closed 2019-11-14 by jeremybennett