embench/embench-iot #10: Benchmarks should not use assert

jeremybennett, 2019-08-29:
Jon Taylor pointed out that assert is not appropriate for benchmarks. See the full mailing list discussion here:

https://lists.librecores.org/pipermail/embench/2019-August/000007.html

I consider this a bug, and this issue is to track a fix to the problem.

jeremybennett, 2019-08-29:
Pull request to fix this in https://github.com/embench/embench-iot/pull/11.

jeremybennett, 2019-11-14:
Marking as closed since this has now been fixed.

closed 2019-11-14 by jeremybennett

fanghuaqi, 2024-01-24:
Hi @jeremybennett , according the benchmark code should avoid using assert rule, for this line https://github.com/embench/embench-iot/blob/master/support/beebsc.c#L65 

`  assert(heap_size % sizeof(void *) == 0);  /* see #138 */`  may need to change to   `assert_beebs(heap_size % sizeof(void *) == 0);  /* see #138 #10  *//`