embench/embench-iot #30: simple primes benchmark

brucehoult, 2019-12-12:
I have found the program at the below address useful for about the last four years as a benchmark of machines ranging from ATmega2560, various ARM devices including M3, M7, A7, A9, A15, A53, A72, various RISC-V, and x86.

On ILP32 or LP64 machines it requires 8004 bytes of globals plus 16 bytes of stack for countPrimes() and 16 or 32 bytes of stack for main(), plus whatever the runtime library needs for printf() and before main().

I hereby submit it for your consideration.

MIT license.

http://hoult.org/primes.txt

jeremybennett, 2020-04-30:
@brucehoult I realize this has been sitting here since the summit without any feedback. Thank you for the submission. I think this is the first candidate for Embench 2.0, when we referesh all the benchmarks to defeat gaming.

PaoloS02, 2024-03-04:
This has been integrated as the primecount benchmark. Closing this now.

closed 2024-03-04 by PaoloS02