embench/embench-iot #31: Error running benchmark

Sameeranjoshi, 2019-12-29:
$ `python build_all.py --arch=/home/odo/benchmarks/embench-iot/config/riscv32/arch.cfg`
  File "build_all.py", line 153
    log.error(f'ERROR: Architecture "{args.arch}" not found: exiting')
                                                                    ^
SyntaxError: invalid syntax

How do I specify the config file.

neilemorgan, 2019-12-30:
Have you tried the following: $ python build_all.py --arch=riscv32 

I'm trying to build for a vegaboard and am using:

python build_all.py --arch=riscv32 --chip=size-test-gcc --board=vegaboard

The main parameter settings in this case are in chip.cfg

Sameeranjoshi, 2019-12-30:
Yes I have tried both, but none works, and fails with same error.
I cloned source , created a 'bd' directory and am running benchmark.
Is there something I am missing?

reSHARMA, 2020-01-19:
You should run the scripts with python3 

```sh
python3 build_all.py --arch riscv32 --chip speed-test --board ri5cyverilator
```

jeremybennett, 2020-01-20:
Thanks @reSHARMA - this is correct. It needs to be Python 3.6 or later. I am working on new versions of the scripts which will check the version of Python. Let me know if this solves your problem.

Sameeranjoshi, 2020-01-25:
Thanks @reSHARMA and @jeremybennett, it solves the problem of running code size benchmarks.

```
aha-mont64
crc32
cubic
edn
huffbench
matmult-int
minver
nbody
nettle-aes
nettle-sha256
nsichneu
picojpeg
qrduino
sglib-combined
slre
st
statemate
ud
wikisort
All benchmarks built successfully
```
Whereas running for speed, I am facing following issue,
I tried web search for the module but couldn't get the solution.
```
./benchmark_speed.py --absolute \
    --target-module run-gdbserver-sim \
    --gdb-command riscv32-unknown-elf-gdb \
    --gdbserver-command riscv32-gdbserver \
    --gdbserver-target ri5cy
**ERROR: Target module import failure: No module named 'run-gdbserver-sim': exiting**
```

Can you point me how to solve the above issue?

jeremybennett, 2020-01-27:
@Sameeranjoshi The target modules are in the `pylib` subdirectory. This is the purpose of the call to `sys.path.append` in `benchmark-speed.py`. For some reason Python is not successfully looking in this directory. You could try adding the `pylib` directory explicitly to your Python module search path.

HTH, Jeremy

jeremybennett, 2020-04-30:
In the absence of further comment, I am marking as closed.

closed 2020-04-30 by jeremybennett