embench/embench-iot #37: Empty Executable Files After Successful Compilation

lyellread, 2020-01-31:
I have been porting `embench-iot` for some custom RISC-V 32 chips, in [this repo](https://github.com/lyellread/embench-iot). In all, I am compiling with the following parameters: 
```
cflags               : ['-O0', '-g', '-march=rv32im', '-mabi=ilp32', '-nostdlib', '-nostartfiles', '-I/home/galoisuser/gfe/benchmarks/embench-iot/support', '-I/home/galoisuser/gfe/benchmarks/embench-iot/config/riscv32/boards/p1-fpga', '-I/home/galoisuser/gfe/benchmarks/embench-iot/config/riscv32/chips/p1', '-I/home/galoisuser/gfe/benchmarks/embench-iot/config/riscv32', '-DCPU_MHZ=50', '-DWARMUP_HEAT=5']
ldflags              : ['-nostartfiles', '-nostdlib', '-march=rv32im', '-mabi=ilp32', '-lc', '-lgcc']
cc                   : riscv64-unknown-elf-gcc
cc_define1_pattern   : -D{0}
cc_define2_pattern   : -D{0}={1}
cc_incdir_pattern    : -I{0}
cc_input_pattern     : /home/galoisuser/gfe/benchmarks/embench-iot/support/dummy-libc.c
cc_output_pattern    : -o {0}
ld_input_pattern     : /home/galoisuser/gfe/benchmarks/embench-iot/support/link.ld
ld_output_pattern    : -o {0}
user_libs            : {}
dummy_libs           : {}
cpu_mhz              : 50
warmup_heat          : 5
ld                   : riscv64-unknown-elf-gcc
```

A couple of notes: 
 - I force-added the `dummy-libc.c` to the `cc_input_pattern` as if I add it to `dummy_libs`, it fails to compile or link. 
 - With the above parameters, compilation and linking succeeds, but all the files have undefined `_start` address, as well as many empty sections:

```
riscv64-unknown-elf-objdump -x ../build/src/aha-mont64/aha-mont64

../build/src/aha-mont64/aha-mont64:     file format elf32-littleriscv
../build/src/aha-mont64/aha-mont64
architecture: riscv:rv32, flags 0x00000112:
EXEC_P, HAS_SYMS, D_PAGED
start address 0xc0000000

Program Header:
    LOAD off    0x00000000 vaddr 0xc007f000 paddr 0xc007f000 align 2**12
         filesz 0x00000054 memsz 0x00281000 flags rw-

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000000  c0000000  c0000000  00000054  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .data         00000000  c0080000  c007f054  00000054  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00000000  c0080000  c0080000  00000000  2**0
                  ALLOC
  3 .stack        00140000  c0080000  c0080000  00001000  2**0
                  ALLOC
  4 .heap         00140000  c01c0000  c01c0000  00001000  2**0
                  ALLOC
SYMBOL TABLE:
c0000000 l    d  .text  00000000 .text
c0080000 l    d  .data  00000000 .data
c0080000 l    d  .bss   00000000 .bss
c0080000 l    d  .stack 00000000 .stack
c01c0000 l    d  .heap  00000000 .heap
c0080800 g       *ABS*  00000000 __global_pointer$
c0080000 g       .bss   00000000 __sbss_start
c0080800 g       .bss   00000000 _gp
c0080000 g       .bss   00000000 __SDATA_BEGIN__
c0080000 g       .data  00000000 __rodata_end
c01c0000 g       .stack 00000000 __freertos_irq_stack_top
c0300000 g       .heap  00000000 _heap_end
c0080000 g       .bss   00000000 __bss_end
c02c0000 g       .heap  00000000 _heap_start
c0080000 g       .bss   00000000 __sbss_end
00000000         *UND*  00000000 _start
c0080000 g       .data  00000000 __rodata_start
c0080000 g       .data  00000000 __data_end
c0080000 g       *ABS*  00000000 __BSS_END__
c0080000 g       .bss   00000000 __bss_start
00040000 g       *ABS*  00000000 _STACK_SIZE
00040000 g       *ABS*  00000000 _HEAP_SIZE
c0180000 g       .stack 00000000 _stack_end
c0080000 g       .data  00000000 __DATA_BEGIN__
c01c0000 g       .stack 00000000 __stack
c0080000 g       *ABS*  00000000 _edata
c0300000 g       .heap  00000000 _end
c02c0000 g       .heap  00000000 _heap
c01c0000 g       .stack 00000000 _stack
c0080000 g       .data  00000000 __data_start
```

Any ideas what could be going on here? Any help or ideas are appreciated. 

~Lyell

jeremybennett, 2020-02-16:
@lyellread Sorry for the slow reply. Did you manage to solve this issue. If not I'll see if I can reproduce it.

lyellread, 2020-02-17:
I was not able to solve it... I am still poking around. I would greatly appreciate if you could take a peek and see if there's something I am doing wrong.