embench/embench-iot #39: Location/repository for board support files

GregAC, 2020-02-18:
I've had Embench running on Ibex/OpenTitan for a while now but in special branches and I'm looking at getting the necessary stuff merged into their respective master branches.

The major question is which repository should the board support files live in? With the current embench build system the board.cfg and boardsupport.c/boardsupport.h at least needs to live in config/riscv32/boards/<board_name> which is in the embench repository. Do you want all such boardsupport files upstreaming into the embench repository?

My preference would be for such files to live in the Ibex/OpenTitan repositories as it makes maintenance simpler (no upstreaming them into this repository). I can hack my way around the build system already (using the vendoring system we have in Ibex/OpenTitan) to make this happen but it would be neater if the build system supported it properly.

This would involve adding an option to the python build system to specify a custom boards/ directory. I am happy to do this but first wanted to ask what the preferred option is here, I see it as either:

* all board support files to be upstreamed into embench-iot
* alter build system as above to support custom board/chip directories

jeremybennett, 2020-02-19:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

> I've had Embench running on Ibex/OpenTitan for a while now but in
> special branches and I'm looking at getting the necessary stuff
> merged into their respective master branches.

Hi Greg,

That's great news.

> The major question is which repository should the board support
> files live in? With the current embench build system the board.cfg
> and boardsupport.c/boardsupport.h at least needs to live in [...]

The key requirement is that others can validate your results. I don't
think that means they have to be in Embench, but they should be freely
available for the lifetime of Embench. There is an issue of making
sure that version control is appropriate - if outside Embench you
probably want to tag to correspond to the versions you used for the
published benchmark result.

I am in the process of cleaning up the current system in time for
Embedded World. In practice only the board support directory really
matters (as per your experience), and all config data is supplied on
the command line, so I'm looking to simplify the config directory to
just architecture and boards and getting rid of the .cfg files
completely. I'm aiming to have that in time for the launch at Embedded
World next week - keep an eye on the pull requests.

> My preference would be for such files to live in the
> Ibex/OpenTitan repositories as it makes maintenance simpler (no
> upstreaming them into this repository). I can hack my way around
> the build system [...]

I'm happy with alternative board support directories, so long as we
can meet the reproducibility criteria. I suggest you wait for my patch
first, to avoid duplicating work, but I would then welcome your patch.

I'm not sure if you have been tracking the discussions from the
monthly calls, but we are changing the baseline to be Arm Cortex M4.
We have the results for this, but we are holding them back for the
launch next week, so journalists can have the first announcement. I
can share them with you if that helps, so long as you don't push
anything using them until next week.

Best wishes,

Jeremy

<https://github.com/embench/embench-iot/issues/39?email_source=notifications&email_token=AAMVYRMT6BBI36X6GHFYQBLRDPHNVA5CNFSM4KXDORA2YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IOJR4HQ>,
or unsubscribe
<https://github.com/
- -- 
Cell: +44 (7970) 676050
SkypeID: jeremybennett
Twitter: @jeremypbennett
Email: jeremy.bennett@embecosm.com
Web: www.embecosm.com
PGP key: 1024D/BEF58172FB4754E1 2009-03-20
-----BEGIN PGP SIGNATURE-----

iF0EARECAB0WIQRASGDWqmhRZUfAaPW+9YFy+0dU4QUCXk0OegAKCRC+9YFy+0dU
4Ud9AJsHySv24OHi9D/MRxTJ3HftI1PlyQCfVjNczycD6+5MTcESNU/yMBcv+oA=
=K8ML
-----END PGP SIGNATURE-----

PaoloS02, 2024-03-04:
I'll close this for now @GregAC as it's been quiet for a while. Happy to reopen it if necessary.

closed 2024-03-04 by PaoloS02