embench/embench-iot #40: Size of multiple sections with same name are reported correctly by python elftools

Tilakbg90, 2020-02-19:
I have been running Embench on AVR with Microchip XC8 and I see incorrect size reported by benchmark_size.py.

.elf file created by Microchip XC8 has multiple sections with same name. 
example of objdump from attached .elf file: 

> 
```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000000  00800200  00800200  0000173a  2**0
                  ALLOC, LOAD, DATA
  1 .text         0000014c  00000000  00000000  00000074  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  7 .text         00000004  000016b8  000016b8  0000172c  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  8 .note.gnu.avr.deviceinfo 00000040  00000000  00000000  00003424  2**2
                  CONTENTS, READONLY, DEBUGGING
  9 .text.mulul64 00000180  00000c06  00000c06  00000c7a  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
 10 .text.montmul 0000022a  000009dc  000009dc  00000a50  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
```

> 

To accommodate all the executable sections I made following changes to benchmark_size.py:

```
def section_relevant_for_size(section):
    flags = section['sh_flags']
    if (flags & SH_FLAGS.SHF_ALLOC) == 0:
        return False;
    if (flags & SH_FLAGS.SHF_EXECINSTR) == 0:
        return False;
    return (flags & SH_FLAGS.SHF_MASKOS) == 0

def benchmark_size(bench):
        for sec in elf.iter_sections():
            if section_relevant_for_size(sec):
                sec_size += sec['sh_size']
```

But this change did not suffice to accommodate the multiple ".text" sections of .elf file.

Total of executable code (without .rodata) of aha-mont64.elf file is 5830 but size reported by benchmark_size.py is 5498. The difference of 332 bytes is from second .text section of size 14C (332 dec)

On further digging I see that "sec['sh_size']" reports only the size of lowest of two .text section. 
.text section of size 14C is not in included in calculation.

You can pass attached .elf file with python elftool to analyze the report.

[aha-mont64.zip](https://github.com/embench/embench-iot/files/4223025/aha-mont64.zip)

jeremybennett, 2020-02-19:
Hi @Tilakbg90 

Thanks for reporting this. Looks like a bug - we are only picking up a single `.text` section. I shall investigate.

If you find a solution and wish to provide a pull request to fix it, that would be welcome.

Best wishes,

Jeremy

PaoloS02, 2020-04-30:
Hello @Tilakbg90,
Thank you for the input.
I added a fix to count all such multiple sections for the size measurement.
I would mark this issue as closed.

Best wishes.
Paolo

closed 2020-04-30 by jeremybennett