embench/embench-iot #51: Problems with spaces when setting tool flag options

MarkHillHuawei, 2020-03-09:
I have noticed that if a user puts extra spaces in their tool options, for example, --cflags “-Os<space><space>–mtune=size” instead of --cflags “-Os<space>–mtune=size” the build fails complaining about a non existent file. This can be a bit confusing, it certainly confused me until I realized what was going on.   This is  due to the split(sep=’ ‘) in build_all.py creating empty elements in the cflags list e.g. [ ‘-Os’, ‘’, ‘-mtune=size’ ].

I think all the split(sep=' ') commands just need to be split() which ignores multiple spaces by default.

jeremybennett, 2020-03-09:
Hi @MarkHillHuawei ,

Thanks for raising this. I'll inviestigate.

Best wishes,

Jeremy

PaoloS02, 2020-04-20:
Thank you for the tip @MarkHillHuawei ,
fixed that now.

Kind regards,
Paolo

PaoloS02, 2020-04-30:
The fix is on master. I would close this issue.

Thank you again.
Paolo

closed 2020-04-30 by jeremybennett