embench/embench-iot #62: Recreating Reference Platform Size Results

joeski65, 2020-05-27:
Hi,

I am trying to recreate the reference platform scores as reported here: https://github.com/embench/embench-iot/tree/master/doc#reference-platform

I fetched the AArch32 bare-metal target GCC 9.2 found [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads).

This comes with binutils 2.33.1 (same as reference platform), and newlib nano 3.1.0 (not the same as reference platform). 

To get the newlib 3.3.0, I cloned the newlib repo and checked out the commit with the tag 'newlib-3.3.0'. I then configured and built using these options (that from my understanding, build 'newlib-nano'):

```./configure --target=arm-none-eabi --enable-newlib-reent-small --disable-newlib-fvwrite-in-streamio --disable-newlib-fseek-optimization --disable-newlib-wide-orient --enable-newlib-nano-malloc --disable-newlib-unbuf-stream-opt --enable-lite-exit --enable-newlib-global-atexit --enable-newlib-nano-formatted-io --disable-nls```

Is this the correct way to build newlib-nano? The README was not explicit as to what options were used to build newlib nano. 

My embench-iot/config/arm/chips/cortex-m4/chip.cfg looks like this:
```
cc = 'arm-none-eabi-gcc'
cflags = [
     '-c', '-Os', '-march=armv7-m', '-mcpu=cortex-m4', '-ffunction-sections', '-mfloat-abi=soft', '-mthumb'
]
ldflags = [
    '-T/path/to/embench-iot/config/arm/boards/stm32f4-discovery/STM32F407XG.ld',
    '-O2', '-Wl,-gc-sections', '-march=armv7-m', '-mcpu=cortex-m4', '-mfloat-abi=soft', '-mthumb', '-specs=nosys.specs'
]
```
and my build_all.py line like this:
`./build_all.py --builddir build_arm_gcc_size --arch arm --chip cortex-m4 --board stm32f4-discovery`

However, when running the `benchmark_size.py`, I get scores that are much larger than 1.00 (upwards of 20!). What am I doing wrong here? Everything seems to fall in line in what's outlined in the README, yet I'm getting such different results.

Thank you,
Joe

jeremybennett, 2020-06-14:
@joeski65 

Apologies for the slow reply. Reproducibility is a key issue. If you look at the latest version of the [embench-iot-results](https://github.com/embench/embench-iot-results), you will see the exact configurations used are now specified for each run.

While we try to keep C library code to a minimum, these are real code kernels, and so will use some library code.  For measuring code size we substitute dummy libraries. This is not ideal, and for Embench 0.6, we will switch to a different method.  Your feedback is much appreciated.  I would encourage you to join the mailing list to discuss (see [embench.org](https://www.embench.org/) and join the video conference held on the third Monday of each Month at 08:00 California time.  The next one is tomorrow, 15 June, agenda sent out very soon.

closed 2020-06-14 by jeremybennett

mistergomes, 2020-12-29:
Hi!

I am actually also trying to reproduce the benchmarks scores for the reference platform.
After configuring the **chip.cfg** file and running the build command:

`$./build_all.py  --arch arm --chip cortex-m4 --board stm32f4-discovery`

I got the following output:

**aha-mont64
crc32
Warning: Link of benchmark "cubic" failed
edn
huffbench
matmult-int
minver
Warning: Link of benchmark "nbody" failed
nettle-aes
nettle-sha256
nsichneu
picojpeg
qrduino
sglib-combined
slre
Warning: Link of benchmark "st" failed
statemate
ud
Warning: Link of benchmark "wikisort" failed**

It seams that for some tests the binary files are not being generated. maybe due to linker problems?
Also, `dding the '-v' to the build command doesn't produce any extra output.

Any ideas on this?

br,

PaoloS02, 2021-01-17:
Hi @mistergomes,
sorry for the slow reply.
I tried to build now the results and it worked just fine.
Could you please share the options that you're using in the chip.cfg file?
Also are you using the binary of the gcc arm downloaded from [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads) like in the previous comments? Or a different source? (this is merely to help me reproduce your issue).
And lastly, just in case this solves the issue, have you added the -lm library?

Best,
Paolo

mistergomes, 2021-01-21:
Hi @PaoloS02 

thanks for your reply.
Yes, the -lm library was missing and the -v produces output to the log file, which I was not looking properly.
Nor everything is working. That was a beginner mistake :)

Best Regards,
Tiago