embench/embench-iot #72: wikisort benchmark uses gcc extensions

kjetilos, 2020-07-07:
The wikisort benchmark is using the __typeof__ gcc extension which will make the benchmark fail on non-gcc compilers like for instance the IAR compiler.

One possible fix is to add the type as an argument to the Var and Swap macros instead of using __typeof__ inside these macros. __typeof__ is also used in the benchmark_body() function and can be replaced by writing the full type.

jeremybennett, 2020-07-19:
Hi @kjetilos 

Thanks for the bug report and sorry for the slow reply. The benchmarks are all supposed to be free of GCC extensions. Assigned to Paolo Savini to do two things

1. Fix this particular issue
2. Run all benchmarks checking for pedantic standards compliance.

Will also raise at our next meeting to get agreement on what C standard we require. Do we really want to be going back to C89 in 2020?

kjetilos, 2020-07-21:
I think it would be ok to require C99, as many compilers support this.