embench/embench-iot #76: Parameter value priority order

Roger-Shepherd, 2020-07-21:
This concerns two points about parameter handling that I'm having a little trouble with understanding.

1. In config/native/chips/size-test-gcc it says

# Parameter values which are duplicated in architecture, board, chip or
# command line are used in the following order of priority
# - default value
# - architecture specific value
# - chip specific value
# - board specific value
# - command line value

I assume this list is in order of increasing priority - that is the command line overrides anything else. I expected the list to be in descending order of priority (highest priority first); unless the order used is absolutely industry standard, and I'm just showing my lack of knowledge, perhaps the intro should say 

# Parameter values which are duplicated in architecture, board, chip or
# command line are used in the following order of priority (lowest first)

2. I can't see how to pass flags through on the command line. Perhaps the mechanism, or lack thereof could be documented.

Roger-Shepherd, 2020-07-21:
Having reread the User Guide in the light of making this comment, I see the guide is perfectly clear on this point. Perhaps the wording from the guid could be used here?

jeremybennett, 2020-08-09:
The way parameters are put together is a legacy of BEEBS. In practice only a few parameters are set in these files and the significant ones are set on the command line. It would be good to update the documentation, but even better would be to simplify the configuration files - perhaps a flat set of files on a per architecture basisi.