embench/embench-iot #78: User guide disagrees with build_all.py script

Roger-Shepherd, 2020-07-21:
The user guide says:

--chip: This mandatory argument ...
--board: This mandatory argument...

In build_all.py they are not mandatory and have the default value "default"

Roger-Shepherd, 2021-04-02:
9 months on and not a single comment on this!

It's easy to fix this either way.

I suspect that changing the script may break some people's builds. Any build relying on the current behaviour of the script will generate an error. 

I propose changing the documentation to match the script.

indumathiraju, 2022-03-16:
Is Embench working on --arch Riscv64

jeremybennett, 2022-03-16:
@Roger-Shepherd Just realized I haven't paid attention to this. Please do update the documentation!

@indumathiraju I'm not aware of any work on 64-bit RISC-V.  Embench IoT is aimed at small microcontrollers, but there is no reason why it cannot be made to work on 64-bit cores.