embench/embench-iot #79: Python version selection

nidalf, 2020-07-22:
**#!/usr/bin/env python3** at the head of the python script will pick **python3.5** if it does exist in the system.
therefore using the following **#/usr/bin/env python3.6** will make sure to pick the right python version

jeremybennett, 2020-07-22:
Hi Nidal,

Thanks for the contribution.

Good point. But what if the user has Python 3.7? I think we have to leave it as python3 and then use the test before executing to check it is at least 3.6.

(BTW - I think there may be a bug in that test to be fixed).

nidalf, 2020-07-23:
What i saw is that the python failed even before getting to the test.
when you have python version less than 3.6 the interpreter itself fails because you are using the f-strings, and this has been introduced in python 3.6, therefore you get a syntax error which happens way before the test starts.
maybe it is possible to convert all the f-string to other format which is acceptable by lower versions of python and this way we insure the script will no break before running the test.

jeremybennett, 2020-07-23:
Hi Nidal - that's the bug that needs fixing. Should not use f strings. That bit of code should work with any version of Python 3. A patch for that would be welcome.

nidalf, 2020-07-23:
I'm closing this issue as it has been fixed and merged

closed 2020-07-23 by nidalf