embench/embench-iot #89: Getting started issue: Unable to build benchmarks

nagendragd, 2020-08-14:
I might have a silly error. I cloned the latest repo (from 8/14). Changed nothing by way of provided code and config. Tried building with:

python3 build_all.py  --builddir . --logdir ./log --arch riscv32 --chip generic  --board ri5cyverilator

Looked into the log which reported this: 
...
General log
===========
Warning: Compilation of beebsc.c from source directory /home/UNT/ndg0068/embench-iot/support to binary directory /home/UNT/ndg0068/embench-iot/./support failed
Command was:
riscv32-unknown-elf-gcc -I/home/UNT/ndg0068/embench-iot/support -I/home/UNT/ndg0068/embench-iot/config/riscv32/boards/ri5cyverilator -I/home/UNT/ndg0068/embench-iot/config/riscv32/chips/generic -I/home/UNT/ndg0068/embench-iot/config/riscv32 -DCPU_MHZ=1 -DWARMUP_HEAT=1 -o beebsc.o /home/UNT/ndg0068/embench-iot/support/beebsc.c

This is not the correct commandline since the build of beebsc.c must use the '-c' flag to tell it only to compile and not to link.

But I could not find a way to pass the '-c' flag. Adding the --cflags -c does not do it. I tried it and it said:
python3 build_all.py  --builddir . --logdir ./log --arch riscv32 --chip generic  --board ri5cyverilator  --cflags -c
usage: build_all.py [-h] [--builddir BUILDDIR] [--logdir LOGDIR] --arch ARCH
                    [--chip CHIP] [--board BOARD] [--cc CC] [--ld LD]
                    [--cflags CFLAGS] [--ldflags LDFLAGS] [--env ENV]
                    [--cc-define1-pattern CC_DEFINE1_PATTERN]
                    [--cc-define2-pattern CC_DEFINE2_PATTERN]
                    [--cc-incdir-pattern CC_INCDIR_PATTERN]
                    [--cc-input-pattern CC_INPUT_PATTERN]
                    [--cc-output-pattern CC_OUTPUT_PATTERN]
                    [--ld-input-pattern LD_INPUT_PATTERN]
                    [--ld-output-pattern LD_OUTPUT_PATTERN]
                    [--user-libs USER_LIBS] [--dummy-libs DUMMY_LIBS]
                    [--cpu-mhz CPU_MHZ] [--warmup-heat WARMUP_HEAT] [-v]
                    [--clean] [--timeout TIMEOUT]
build_all.py: error: argument --cflags: expected one argument

Does the invocation require the '-c' compiler flag to be passed? If so, it seems that could be included in the build step automatically.

And if the flag has to be passed, then how exactly should the --cflags options be used?

Thanks
Nagendra

sobuch, 2020-08-14:
You can use `--cflags="-c"`, discussion here may be helpful when starting https://github.com/embench/embench-iot/issues/2.

nagendragd, 2020-08-15:
Yes thank you, that was a silly issue. Thanks.

closed 2020-08-15 by nagendragd