embench/embench-iot #93: Standard Citation Template

ben-marshall, 2020-09-11:
Hi there

Is it possible to come up with a standardised citation template for Embench? Those of us who reference it in academic publications can create an ad-hoc citation fairly easily, but a consistent one we can copy and paste into our bibliographies would be super helpful.

Currently, my Bibtex entry looks like:
```
@misc{embench-iot,
title={Embench: Open Benchmarks for Embedded Platforms},
author={David Patterson and Jeremy Bennett and Palmer Dabbelt, Cesare Garlati and G. S. Madhusudan and Trevor Mudge},
howpublished="\url{https://github.com/embench/embench-iot}"
}
```
Is this suitable, and could something like it be pasted on the front page for everyone to use?

Cheers,
Ben

jeremybennett, 2020-09-12:
Hi Ben - good question. I'll put it on the agenda for the meeting on 21 September. Will you be able to attend?

ben-marshall, 2020-09-14:
Hi Jeremy,
I'll do my best to attend for sure. Where can I find the details?
Also - I think the website link on the top right of the repo homepage might be wrong: https://www.embench.org -> https://embench.org
Cheers,
Ben

jeremybennett, 2020-09-14:
Thanks - I have sent you an invite (actually sent you two - I invited you to the wrong meeting first). I'll get the website fixed - thanks for spotting.