embench/embench-iot #95: Does Wikisort need to use double precision constants ?

axiomlegend, 2020-09-21:
It seems that this benchmark is measuring floating point more than wiki sorting algorithms.
Using 1.0f, 2.5f, 0.9f in the Testing* routines would be the recommend change.

Thanks.

Roger-Shepherd, 2020-09-22:
From a quick glance it looks like double v. single precision f.p. would make essentially zero difference in the execution time. There might be an impact on code size but unless (i) the difference was significant, and (ii) the use of double precision could be considered "unreasonable" I'd leave it alone. Rewriting code in benchmarks, particularly, "real-world" benchmarks is a slippery slope.

jeremybennett, 2020-09-22:
@axiomlegend One of the principles on which Embench is based is that code should be "real programs" - warts and all.  The only changes we make a i) to ensure they meet strict standard C11, ii) they work correctly on everything from 8- to 64-bit architectures and iii) we ensure they self-check.

Almost all could be written much better, but we didn't want a collection of perfect programs, just what ordinary programmers create.

The program suite is refreshed every two years, so please do submit candidate programs for future versions of the benchmark suite.