embench/embench-iot #96: Explanation / Origin of benchmarks

nagendragd, 2020-10-01:
Hello,

I am looking for an overview / explanation of what the benchmarks do, why were they picked, etc.
Tried looking at the mentioned link: http://beebs.eu/ and it seems to redirect to a bad site.

Info on these benchmarks will be very useful.

thanks
Nagendra

jeremybennett, 2020-10-05:
Hi @nagendragd 

Sorry about your experience with this. We discovered last week that we had lost the beebs.eu domain - renewal emails had been misdirected.

BEEBS will shortly reappear as beebs.mageec.org, since it was developed as part of the MAGEEC research project funded by the UK government.  In the meantime, you can find the source on GitHub: https://github.com/mageec/beebs.  This gives you two links to papers, available on arXiv:

- [BEEBS: Open Benchmarks for Energy Measurements on Embedded Platforms](https://arxiv.org/abs/1308.5174) by James Pallister, Simon Hollis and Jeremy Bennett
- [Identifying Compiler Options to Minimise Energy Consumption for Embedded Platforms](https://arxiv.org/abs/1303.6485) also 
by James Pallister, Simon Hollis and Jeremy Bennett

The first describes the original BEEBS benchmarks, the second shows them in use. Since the original paper, the benchmarks have been expanded to 79 programs. I used them in an [analysis of RISC-V code size](https://archive.fosdem.org/2019/schedule/event/riscvcompact/) at FOSDEM 2019.

Hope this is useful.

nagendragd, 2020-10-05:
Ok thanks. Appreciate the direct links.

Best regards
Nagendra

jeremybennett, 2020-10-18:
BEEBS can now be found at https://beebs.mageec.org/. We need to fix the links. Marked as a bug and assigned to me to fix.

jeremybennett, 2020-11-13:
Fixed in commit bae259a. Link is now correct.

closed 2020-11-13 by jeremybennett