embench/embench-iot #97: Issue Running Embench

GeorgeOnwubuya, 2020-10-07:
I tried to compile the benchmarks using the following command:

**_python3 build_all.py --arch arm --chip cortex-m4 --board generic_** 

I am running into the error below
Warning: Compilation of beebsc.c from source directory /Users/user/embench-iot/support to binary directory /Users/user/embench-iot/bd/support failed
Warning: Compilation of main.c from source directory /Users/user/embench-iot/support to binary directory /Users/user/embench-iot/bd/support failed
Warning: Compilation of chipsupport.c from source directory /Users/user/embench-iot/config/arm/chips/cortex-m4 to binary directory /Users/user/embench-iot/bd/config/arm/chips/cortex-m4 failed
Warning: Compilation of boardsupport.c from source directory /Users/user/embench-iot/config/arm/boards/generic to binary directory /Users/user/embench-iot/bd/config/arm/boards/generic failed
Warning: Compilation of mont64.c from source directory /Users/user/embench-iot/src/aha-mont64 to binary directory /Users/user/embench-iot/bd/src/aha-mont64 failed
Warning: Compilation of crc_32.c from source directory /Users/user/embench-iot/src/crc32 to binary directory /Users/user/embench-iot/bd/src/crc32 failed
Warning: Compilation of libcubic.c from source directory /Users/user/embench-iot/src/cubic to binary directory /Users/user/embench-iot/bd/src/cubic failed
Warning: Compilation of basicmath_small.c from source directory /Users/user/embench-iot/src/cubic to binary directory /Users/user/embench-iot/bd/src/cubic failed
Warning: Compilation of libedn.c from source directory /Users/user/embench-iot/src/edn to binary directory /Users/user/embench-iot/bd/src/edn failed
Warning: Compilation of libhuffbench.c from source directory /Users/user/embench-iot/src/huffbench to binary directory /Users/user/embench-iot/bd/src/huffbench failed
Warning: Compilation of matmult-int.c from source directory /Users/user/embench-iot/src/matmult-int to binary directory /Users/user/embench-iot/bd/src/matmult-int failed
Warning: Compilation of libminver.c from source directory /Users/user/embench-iot/src/minver to binary directory /Users/user/embench-iot/bd/src/minver failed
Warning: Compilation of nbody.c from source directory /Users/user/embench-iot/src/nbody to binary directory /Users/user/embench-iot/bd/src/nbody failed
Warning: Compilation of nettle-aes.c from source directory /Users/user/embench-iot/src/nettle-aes to binary directory /Users/user/embench-iot/bd/src/nettle-aes failed
Warning: Compilation of nettle-sha256.c from source directory /Users/user/embench-iot/src/nettle-sha256 to binary directory /Users/user/embench-iot/bd/src/nettle-sha256 failed
Warning: Compilation of libnsichneu.c from source directory /Users/user/embench-iot/src/nsichneu to binary directory /Users/user/embench-iot/bd/src/nsichneu failed
Warning: Compilation of picojpeg_test.c from source directory /Users/user/embench-iot/src/picojpeg to binary directory /Users/user/embench-iot/bd/src/picojpeg failed
Warning: Compilation of libpicojpeg.c from source directory /Users/user/embench-iot/src/picojpeg to binary directory /Users/user/embench-iot/bd/src/picojpeg failed
Warning: Compilation of qrtest.c from source directory /Users/user/embench-iot/src/qrduino to binary directory /Users/user/embench-iot/bd/src/qrduino failed
Warning: Compilation of qrencode.c from source directory /Users/user/embench-iot/src/qrduino to binary directory /Users/user/embench-iot/bd/src/qrduino failed
Warning: Compilation of qrframe.c from source directory /Users/user/embench-iot/src/qrduino to binary directory /Users/user/embench-iot/bd/src/qrduino failed
Warning: Compilation of combined.c from source directory /Users/user/embench-iot/src/sglib-combined to binary directory /Users/user/embench-iot/bd/src/sglib-combined failed
Warning: Compilation of libslre.c from source directory /Users/user/embench-iot/src/slre to binary directory /Users/user/embench-iot/bd/src/slre failed
Warning: Compilation of libst.c from source directory /Users/user/embench-iot/src/st to binary directory /Users/user/embench-iot/bd/src/st failed
Warning: Compilation of libstatemate.c from source directory /Users/user/embench-iot/src/statemate to binary directory /Users/user/embench-iot/bd/src/statemate failed
Warning: Compilation of libud.c from source directory /Users/user/embench-iot/src/ud to binary directory /Users/user/embench-iot/bd/src/ud failed
Warning: Compilation of libwikisort.c from source directory /Users/user/embench-iot/src/wikisort to binary directory /Users/user/embench-iot/bd/src/wikisort failed

Any idea on how to resolve this? 

Thanks,
George O.

Roger-Shepherd, 2020-10-07:
You need to get more error messages to diagnose this. I don't have a gcc installation on my development machine so can't check this. If you look in buildall.py you'll be able to see which flags give you the messages you need. I think you need -v (verbose)>

jeremybennett, 2020-10-07:
@GeorgeOnwubuya There should be a log file created (in the `logs` directory by default I think). If you can post that is should have more information.

GeorgeOnwubuya, 2020-10-07:
I have posted the log file below 

Log file: /Users/user/embench-iot/logs/build-2020-10-06-221935.log

Supplied arguments
==================
--builddir            : bd
--logdir              : logs
--arch                : arm
--chip                : cortex-m4
--board               : generic
--cc                  : None
--ld                  : None
--cflags              : None
--ldflags             : None
--env                 : None
--cc-define1-pattern  : None
--cc-define2-pattern  : None
--cc-incdir-pattern   : None
--cc-input-pattern    : None
--cc-output-pattern   : None
--ld-input-pattern    : None
--ld-output-pattern   : None
--user-libs           : None
--dummy-libs          : None
--cpu-mhz             : None
--warmup-heat         : None
--verbose             : False
--clean               : False
--timeout             : 5

Benchmarks
==========
aha-mont64
crc32
cubic
edn
huffbench
matmult-int
minver
nbody
nettle-aes
nettle-sha256
nsichneu
picojpeg
qrduino
sglib-combined
slre
st
statemate
ud
wikisort

Global parameters
=================
rootdir              : /Users/user/embench-iot
verbose              : False
bd                   : /Users/user/embench-iot/bd
configdir            : /Users/user/embench-iot/config
bd_configdir         : /Users/user/embench-iot/bd/config
archdir              : /Users/user/embench-iot/config/arm
bd_archdir           : /Users/user/embench-iot/bd/config/arm
chipdir              : /Users/user/embench-iot/config/arm/chips/cortex-m4
bd_chipdir           : /Users/user/embench-iot/bd/config/arm/chips/cortex-m4
boarddir             : /Users/user/embench-iot/config/arm/boards/generic
bd_boarddir          : /Users/user/embench-iot/bd/config/arm/boards/generic
env                  : {}
benchdir             : /Users/user/embench-iot/src
bd_benchdir          : /Users/user/embench-iot/bd/src
supportdir           : /Users/user/embench-iot/support
[... 387 lines cut ...]
collect2: error: ld returned 1 exit status

Warning: Compilation of libwikisort.c from source directory /Users/user/embench-iot/src/wikisort to binary directory /Users/user/embench-iot/bd/src/wikisort failed
Command was:
arm-none-eabi-gcc -I/Users/user/embench-iot/support -I/Users/user/embench-iot/config/arm/boards/generic -I/Users/user/embench-iot/config/arm/chips/cortex-m4 -I/Users/user/embench-iot/config/arm -DCPU_MHZ=1 -DWARMUP_HEAT=1 -o libwikisort.o /Users/user/embench-iot/src/wikisort/libwikisort.c

/Users/user/Local/gcc_arm/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
/Users/user/Local/gcc_arm/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/lib/crt0.o: In function `start':
(.text+0xec): undefined reference to `main'
/var/folders/mz/l0j9fwqx5d51h7gyx621lbz80000gn/T//cczyq3OR.o: In function `WikiSort':
libwikisort.c:(.text+0xf50): undefined reference to `sqrt'
/var/folders/mz/l0j9fwqx5d51h7gyx621lbz80000gn/T//cczyq3OR.o: In function `TestingRandom':
libwikisort.c:(.text+0x3f00): undefined reference to `rand_beebs'
/var/folders/mz/l0j9fwqx5d51h7gyx621lbz80000gn/T//cczyq3OR.o: In function `TestingMostlyDescending':
libwikisort.c:(.text+0x3f48): undefined reference to `rand_beebs'
/var/folders/mz/l0j9fwqx5d51h7gyx621lbz80000gn/T//cczyq3OR.o: In function `TestingMostlyAscending':
libwikisort.c:(.text+0x4024): undefined reference to `rand_beebs'
/var/folders/mz/l0j9fwqx5d51h7gyx621lbz80000gn/T//cczyq3OR.o: In function `TestingJittered':
libwikisort.c:(.text+0x4170): undefined reference to `rand_beebs'
/var/folders/mz/l0j9fwqx5d51h7gyx621lbz80000gn/T//cczyq3OR.o: In function `TestingMostlyEqual':
libwikisort.c:(.text+0x420c): undefined reference to `rand_beebs'
/var/folders/mz/l0j9fwqx5d51h7gyx621lbz80000gn/T//cczyq3OR.o: In function `benchmark_body':
libwikisort.c:(.text+0x4378): undefined reference to `srand_beebs'
collect2: error: ld returned 1 exit status

jeremybennett, 2020-10-07:
@GeorgeOnwubuya OK - I see the issue. For GCC you need the `-c` flag to tell it to build an object file and not link. So you need to add `--cflags=-c` to your command line.

HTH

GeorgeOnwubuya, 2020-10-07:
I added the --cflags=-c but I am still running into errors. I have posted the log file below

Log file: /Users/user/embench-iot/logs/build-2020-10-07-092130.log

Supplied arguments
==================
--builddir            : bd
--logdir              : logs
--arch                : arm
--chip                : cortex-m4
--board               : generic
--cc                  : None
--ld                  : None
--cflags              : -c
--ldflags             : None
--env                 : None
--cc-define1-pattern  : None
--cc-define2-pattern  : None
--cc-incdir-pattern   : None
--cc-input-pattern    : None
--cc-output-pattern   : None
--ld-input-pattern    : None
--ld-output-pattern   : None
--user-libs           : None
--dummy-libs          : None
--cpu-mhz             : None
--warmup-heat         : None
--verbose             : False
--clean               : False
--timeout             : 5

Benchmarks
==========
aha-mont64
crc32
cubic
edn
huffbench
matmult-int
minver
nbody
nettle-aes
nettle-sha256
nsichneu
picojpeg
qrduino
sglib-combined
slre
st
statemate
ud
wikisort

Global parameters
=================
rootdir              : /Users/user/embench-iot
verbose              : False
bd                   : /Users/user/embench-iot/bd
configdir            : /Users/user/embench-iot/config
bd_configdir         : /Users/user/embench-iot/bd/config
archdir              : /Users/user/embench-iot/config/arm
bd_archdir           : /Users/user/embench-iot/bd/config/arm
chipdir              : /Users/user/embench-iot/config/arm/chips/cortex-m4
bd_chipdir           : /Users/user/embench-iot/bd/config/arm/chips/cortex-m4
boarddir             : /Users/user/embench-iot/config/arm/boards/generic
bd_boarddir          : /Users/user/embench-iot/bd/config/arm/boards/generic
env                  : {}
benchdir             : /Users/user/embench-iot/src
bd_benchdir          : /Users/user/embench-iot/bd/src
supportdir           : /Users/user/embench-iot/support
[... 199 lines cut ...]
/Users/user/Local/gcc_arm/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2: error: ld returned 1 exit status

Compilation of benchmark "ud" successful
Warning: Link of benchmark "ud" failed
In directory "/Users/user/embench-iot/bd/src/ud"
Command was:
arm-none-eabi-gcc -o ud libud.o /Users/user/embench-iot/bd/config/arm/chips/cortex-m4/chipsupport.o /Users/user/embench-iot/bd/config/arm/boards/generic/boardsupport.o /Users/user/embench-iot/bd/support/main.o /Users/user/embench-iot/bd/support/beebsc.o

/Users/user/Local/gcc_arm/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2: error: ld returned 1 exit status

Compilation of benchmark "wikisort" successful
Warning: Link of benchmark "wikisort" failed
In directory "/Users/user/embench-iot/bd/src/wikisort"
Command was:
arm-none-eabi-gcc -o wikisort libwikisort.o /Users/user/embench-iot/bd/config/arm/chips/cortex-m4/chipsupport.o /Users/user/embench-iot/bd/config/arm/boards/generic/boardsupport.o /Users/user/embench-iot/bd/support/main.o /Users/user/embench-iot/bd/support/beebsc.o

/Users/user/Local/gcc_arm/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
libwikisort.o: In function `WikiSort':
libwikisort.c:(.text+0xf50): undefined reference to `sqrt'
collect2: error: ld returned 1 exit status

jeremybennett, 2020-10-07:
@GeorgeOnwubuya You are now getting link failures, and the errors indicate a problem with your C library setup. Here is the first example of the relevant message
```
/Users/user/Local/gcc_arm/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function exit': exit.c:(.text.exit+0x2c): undefined reference to _exit'
```
So you do have a C library - it's finding `exit` in `libc.a`, but it is not finding `_exit`. This would normally be part of the libgloss board support package which is part of the newlib C library. Do you have a file `libgloss.a` in `/Users/user/Local/gcc_arm/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/../../../../arm-none-eabi/lib`?

I'm assigning to @PaoloS02 since he has most recently built the Arm target, in case there is something else needed for Arm.

GeorgeOnwubuya, 2020-10-07:
It doesn't look like I have the libgloss.a file but I do have libgloss-linux.a file. 
 How do I get this file installed on my end? Is there a package I need to install? 

Thanks for all your help.

PaoloS02, 2020-10-07:
Hello @GeorgeOnwubuya ,
one thing that I see at a first glance is that you need to include the math library. You can use the --user-libs option from the build_all.py script. So you need to add --user-libs="-lm" to your command line.
I'll come back to you as soon as I have checked the issue of the _exit symbol.

PaoloS02, 2020-10-07:
Hello @GeorgeOnwubuya ,
as we are in the context of embedded systems we need to specify how to enable the system calls.
With newlib come several sets of systemcalls that are enabled with the --specs options:
```
arm-none-eabi-gcc --specs=rdimon.specs
arm-none-eabi-gcc --specs=nosys.specs
```
rdimon provides syscalls that rely on semihosting. Nosys instead provides a set of stubs that retarget the system calls.
We usually use nosys.
In this case for instance you need to add the linker option:

```
--ldflags='--specs=nosys.specs'
```

Hope this helps.
Paolo

GeorgeOnwubuya, 2020-10-07:
Thanks for your speedy reply Paolo. 

It looks like I don't have the nosys.specs in my gcc directory. Any thought on how to get this pacakge?

Log file: /Users/user/embench-iot/logs/build-2020-10-07-113903.log

Supplied arguments
==================
--builddir            : bd
--logdir              : logs
--arch                : arm
--chip                : cortex-m4
--board               : generic
--cc                  : None
--ld                  : None
--cflags              : -c
--ldflags             : --specs=nosys.specs
--env                 : None
--cc-define1-pattern  : None
--cc-define2-pattern  : None
--cc-incdir-pattern   : None
--cc-input-pattern    : None
--cc-output-pattern   : None
--ld-input-pattern    : None
--ld-output-pattern   : None
--user-libs           : -lm
--dummy-libs          : None
--cpu-mhz             : None
--warmup-heat         : None
--verbose             : False
--clean               : False
--timeout             : 5

Benchmarks
==========
aha-mont64
crc32
cubic
edn
huffbench
matmult-int
minver
nbody
nettle-aes
nettle-sha256
nsichneu
picojpeg
qrduino
sglib-combined
slre
st
statemate
ud
wikisort

Global parameters
=================
rootdir              : /Users/user/embench-iot
verbose              : False
bd                   : /Users/user/embench-iot/bd
configdir            : /Users/user/embench-iot/config
bd_configdir         : /Users/user/embench-iot/bd/config
archdir              : /Users/user/embench-iot/config/arm
bd_archdir           : /Users/user/embench-iot/bd/config/arm
chipdir              : /Users/user/embench-iot/config/arm/chips/cortex-m4
bd_chipdir           : /Users/user/embench-iot/bd/config/arm/chips/cortex-m4
boarddir             : /Users/user/embench-iot/config/arm/boards/generic
bd_boarddir          : /Users/user/embench-iot/bd/config/arm/boards/generic
env                  : {}
benchdir             : /Users/user/embench-iot/src
[... 149 lines cut ...]
arm-none-eabi-gcc: error: nosys.specs: No such file or directory

Compilation of benchmark "statemate" successful
Warning: Link of benchmark "statemate" failed
In directory "/Users/user/embench-iot/bd/src/statemate"
Command was:
arm-none-eabi-gcc --specs=nosys.specs -o statemate libstatemate.o /Users/user/embench-iot/bd/config/arm/chips/cortex-m4/chipsupport.o /Users/user/embench-iot/bd/config/arm/boards/generic/boardsupport.o /Users/user/embench-iot/bd/support/main.o /Users/user/embench-iot/bd/support/beebsc.o -lm

arm-none-eabi-gcc: error: nosys.specs: No such file or directory

Compilation of benchmark "ud" successful
Warning: Link of benchmark "ud" failed
In directory "/Users/user/embench-iot/bd/src/ud"
Command was:
arm-none-eabi-gcc --specs=nosys.specs -o ud libud.o /Users/user/embench-iot/bd/config/arm/chips/cortex-m4/chipsupport.o /Users/user/embench-iot/bd/config/arm/boards/generic/boardsupport.o /Users/user/embench-iot/bd/support/main.o /Users/user/embench-iot/bd/support/beebsc.o -lm

arm-none-eabi-gcc: error: nosys.specs: No such file or directory

Compilation of benchmark "wikisort" successful
Warning: Link of benchmark "wikisort" failed
In directory "/Users/user/embench-iot/bd/src/wikisort"
Command was:
arm-none-eabi-gcc --specs=nosys.specs -o wikisort libwikisort.o /Users/user/embench-iot/bd/config/arm/chips/cortex-m4/chipsupport.o /Users/user/embench-iot/bd/config/arm/boards/generic/boardsupport.o /Users/user/embench-iot/bd/support/main.o /Users/user/embench-iot/bd/support/beebsc.o -lm

arm-none-eabi-gcc: error: nosys.specs: No such file or directory

PaoloS02, 2020-10-07:
Hello @GeorgeOnwubuya ,

could you please post the information of your toolchain?
```
arm-none-eabi-gcc -v
```

Thanks
Paolo

GeorgeOnwubuya, 2020-10-07:
Information on toolchain: 

Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/Users/user/Local/gcc_arm/gcc-arm-none-eabi-4_7-2013q3/bin/../lib/gcc/arm-none-eabi/4.7.4/lto-wrapper
Target: arm-none-eabi
Configured with: /Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/src/gcc/configure --target=arm-none-eabi --prefix=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/install-native --libexecdir=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/install-native/lib --infodir=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/install-native/arm-none-eabi --build=x86_64-apple-darwin10 --host=x86_64-apple-darwin10 --with-gmp=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/build-native/host-libs/usr --with-mpfr=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/build-native/host-libs/usr --with-mpc=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/build-native/host-libs/usr --with-ppl=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/build-native/host-libs/usr --with-cloog=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/build-native/host-libs/usr --with-libelf=/Users/build/GCC-4-7-build/gcc-arm-none-eabi-4_7-2013q3-20130916/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-lstdc++ -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r
Thread model: single
gcc version 4.7.4 20130913 (release) [ARM/embedded-4_7-branch revision 202601] (GNU Tools for ARM Embedded Processors)

PaoloS02, 2020-10-07:
Hi @GeorgeOnwubuya ,

Ok, you could try to look for nosys under your library folders like /Users/lib/arm-none-eabi. I'm not sure where it should be on your system, but it should contain it. Just if you want to make sure you haven't got it there while maybe defaulting to another installed toolchain by mistake.

Anyway if I'm correct that toolchain has been installed as a packet for your system, right? Usually in order to have a better control on what libraries you are using and what you are including, but above all in order to have control on the configuration of your toolchain and change it if necessary I'd suggest building gcc from scratch. If you want to have a look at the configuration we used for gcc to produce our results you can look here: https://github.com/embench/embench-iot-results
Every set of results has the full configuration for gcc, binutils and newlib.
@jeremybennett have we got somewhere to share the script we used to build our toolchains?

In the meantime you could also try to install a newer version of your arm toolchain (if it was installed as a packet). I've read here: https://trac.macports.org/ticket/57572 that they had a similar problem and fixed it with a patch. They switched then to gcc 9.2.
I'm not sure whether that's where your toolchain comes from, but its a guess.

Best,
Paolo

GeorgeOnwubuya, 2020-10-07:
I had to donwload a more up to date version of gcc-arm-none-eabi. I have sucessfully built the benchmarks: 

**aha-mont64
crc32
cubic
edn
huffbench
matmult-int
minver
nbody
nettle-aes
nettle-sha256
nsichneu
picojpeg
qrduino
sglib-combined
slre
st
statemate
ud
wikisort
All benchmarks built successfully**

I do have a few more questions. 
-> How do I run the benchmarks? I know you can run them with respect to speed and size. I am just curious as to what the minimum configuration looks like. 
-> Is there a way to run a single benchmark at a time?

Thanks, 
George O.

PaoloS02, 2020-10-08:
Hello @GeorgeOnwubuya ,
once you built the tests you need to run benchmark_size.py for size results and benchmark_speed.py for speed results.
If you invoke them without arguments you'll see the required options. The size script doesn't need options, but you might want anyway to have a look at them in order to decide at least whether to have absolute results or results relative to the baseline.

The speed script instead requires that you at least pass a target module. These are the python modules in pylib/. Each of these modules describes a debug session that needs to interface with real hardware or at least emulated hardware (e.g. ri5cy verilated model).

Have a look at the results repo to have an idea of what kind of options we usually use for the tests: https://github.com/embench/embench-iot-results

You'll find also how we ran the baseline that I mentioned earlier, that is the sets of results against which the scripts calculate the relative data: https://github.com/embench/embench-iot-results/blob/master/details/cortexm4-armv7m-gcc-9.2-o2.mediawiki

Hope this helps.
Paolo

GeorgeOnwubuya, 2020-10-08:
Thanks @PaoloS02. 
I ran the benchmark_size.py and I am running into the error pasted below. 

**Traceback (most recent call last):
  File "benchmark_size.py", line 24, in <module>
    from elftools.elf.elffile import ELFFile
ModuleNotFoundError: No module named 'elftools'**

Where do I find the module named 'elftools?

PaoloS02, 2020-10-12:
Hi @GeorgeOnwubuya 

you probably need to install pyelftools if you haven't got it already.
Does this work?
```
pip3 install pyelftools
```

Let me know
Paolo

jeremybennett, 2020-10-18:
@PaoloS02 I've marked as an enhancement, to allow you to update the documentation in the light of @GeorgeOnwubuya experience.