embench/embench-iot #98: Queries on Embench benchmarks

sheenagoel696, 2020-10-16:
Hi,

I want to run embench benchmarks for arm cpu cortex-m55F and i am not sure about board,
could you please let me know what options i need to select?

Should it be arc=arm, chip=cortex-m55f, board=generic?

Any help would be greatly appreciated.

Thankyou.

Regards,
Sheena

jeremybennett, 2020-10-18:
@sheenagoel696 

We don't have experience with running for this processor. You will need to create your own chip and board descriptions, but you can base these on the Cortex M4 and STM Discovery versions. I've assigned to @PaoloS02 who set up the Cortex M4 boards. He may have further advice for you.

PaoloS02, 2020-11-16:
@sheenagoel696 

Sorry for the slow reply. You probably already solved this problem but I'll answer anyway for future reference or in case someone else finds it useful.
Apologies if I'll be obvious but it is just for clarity. The `board` directory should contain compilation options, linker scripts, startup files or other information that optimize the compilation of Embench in relation to the features of the board that hosts the chip.
If you're uncertain then I would start with the default generic board. So you don't need to specify it in the command line. You could try with that and see how it works.

If you want to use some features of your board that improve the performance compared to the standard compiler support for ARM you can copy the `generic` board directory and customize it like we did for `stm32f4-discovery` that contains a specific linker script and startup file.
If instead you're not interested in the features of your board and only want to test Embench on your chip with the 'default' settings you can just use the 'generic' board.

Anyway if you want to create your own startup file and linker script for instance for instance, what I did for stm32f4-discovery was to use the STM software called STM32CubeMX: https://www.st.com/en/development-tools/stm32cubemx.html
that generates the right startup file and linker script according to the board/chip settings you want (an higher frequency oscillator, a specific debug module, a specific clock counter, etc... ). Depending on the MCU you're using you could use the same software if it is an stm or maybe there's a similar one for yours. Otherwise there should be some examples available online for cortex-m55F?

Best,
Paolo