embench/embench-iot #103: Bug in build_all.py

Roger-Shepherd, 2020-11-16:
I've just encountered a bug in build_all.py. I hit a timeout in  If you hit a timeout when linking you get the following messages:

Warning: link of benchmark "aha-mont64" timed out
Traceback (most recent call last):
  File "./build_all.py", line 751, in <module>
    sys.exit(main())
  File "./build_all.py", line 737, in main
    res = link_benchmark(bench)
  File "./build_all.py", line 686, in link_benchmark
    log.debug(res.stdout.decode('utf-8'))
UnboundLocalError: local variable 'res' referenced before assignment

The problem seems to be that when res is used (686) it hasn't been set. In non-timeout operation it would be set on line 667 but the Timeout.Expired exception causes 686 not to complete.

jeremybennett, 2020-11-17:
I thought this had been fixed. Assigned to @PaoloS02 to investigate.

PaoloS02, 2020-11-30:
Hi @Roger-Shepherd, @jeremybennett,
I see that this problem was the same as the one addresses by merge request #15 for the compile_file
It seems that pull request #64 has brought back this issue, also for compilation I think.
The point of pull request #64 is to have debug info even if the compilation/linking has timed out.
We need a compromise to have that, without using undefined/unassigned variables.
I'll look at it as soon as I can, but maybe a python expert will have a better solution here.

jeremybennett, 2021-01-10:
Resolved by PR #105

closed 2021-01-10 by jeremybennett