embench/embench-iot #111: '--json-comma' and '--no-json-comma' options to benchmark scripts are not documented

Roger-Shepherd, 2021-02-17:
Both benchmark_size.py and benchmark_speed.py will take '--json-comma' and '--no-json-comma'. These affect the formatting of json formatted results. This is not documented.

Roger-Shepherd, 2021-03-10:
I'm happy to fix this as suggested as produce the necessary pull request if someone tells me it looks like I'm on the right track.

jeremybennett, 2021-03-14:
Roger - the JSON output was work in progress. It produces fragments to paste into complete files for embench-iot-results. What it should do is produce the complete file, but that requires

1. Finding a way to incorporate all the data about the platform, compiler, etc.
2. Finding a way co coordinate the output from `benchmark-size.py` and `benchmark-speed.py`.

Roger-Shepherd, 2021-03-14:
This make perfect sense now. The job of producing the complete output does require some thought, and perhaps some changes to the way things are done at the moment.

PaoloS02, 2021-03-23:
I'm in the process of adding a note on the `--metric` option to the documentation. I could document `--json-comma` and `--no-json-comma` too, while I'm at it. This if we agreed that we keep such options.

Roger-Shepherd, 2021-03-23:
I think we should leave them undocumented, at least until they are fixed. Once they are fixed, and work so the output can be incorporated into the results file as Jeremy describes, they should be documented. I've not looked at how well they so this so don't know whether it is sufficient just to fix them and document.