embench/embench-iot #114: [Major Need] Proper Documentation of README for CONTRIBUTION

ArchishmanSengupta, 2021-03-10:
embench-iot is indeed a great project but it can be better. As an user & i would like to make **CHANGES** in the `README` file or add a `CONTRIBUTING.md` file so that a developer who wants to **CONTRIBUTE** can easily follow the steps of its installation starting from **Forking the repo**. 

This is will cover Issue:
1. #111 
2. #112 
3. #113 

Please assign me this and please provide a communication medium (slack or any other).

Thank YOU

jeremybennett, 2021-03-14:
Hi @ArchishmanSengupta

All contributions are welcome. I would encourage you to join our monthly calls. The next one is at 08:00 Pacific Time on Monday 15 March (tomorrow).  Agenda and videoconference details will be published shortly on the [Embench mailing list](https://lists.librecores.org/listinfo/embench).

ArchishmanSengupta, 2021-03-14:
will be joining the call. Thanks