embench/embench-iot #118: Why do we benchmark size using .text and not .text and .rodata?

Roger-Shepherd, 2021-03-19:
See issue #117 for some background

Since the program and read only data have to be loaded into memory before execution can occur it would seem sensible that the size of an executable included .rodata as well as .text. This is a particular issue as there are often choices about how (constant) data is loaded - e.g. instructions with literals v. load from initialised memory.

Apologies if this choice is documented, I couldn't find it on GitHub

jeremybennett, 2021-03-19:
There are arguments for a number of variants, which were discussed early on. The `benchmark_size.py` script can actually compute any of these, the default is just `.text`.

* You can look at `.text` + `.rodata` because this goes in ROM
* You can look at `.text`  + `.rodata` + `.data` because the initial values for `.data` also need to go into ROM and be copied out by the startup code.
* You can look at `.text` + `.rodata` + `.data` + `.bss` because this represents *all* the memory the program needs to get started.

We went with `.text` because it is simplest, and most relevant to code optimization. Of course there are optimizations which are data driven and so can trade space in `.text` and the various data segments.

HTH, Jeremy

Roger-Shepherd, 2021-03-19:
Thanks. I can also see the answer to issue #119 coming from this.......

closed 2021-03-19 by Roger-Shepherd