embench/embench-iot #119: bench_speed.py - text, data, rodata, bss and metric parameter handling

Roger-Shepherd, 2021-03-19:
See isse #117 for background

The documentation for the parameters `text`, `data`, `rodata`, `bss` and `metric` all say they take a `space separated list` as in 

>  `--text`: A space separated list of sections containing code.  Default value `.text`.

The implementation uses `action='append'

>     parser.add_argument(
>         '--text',
>         type=str,
>         default=[],
>         action='append',
>         help='Section name(s) containing code'
>     )

which requires the parameter name to be mentioned for each parameter rather than accepting a space separated list of arguments. To get the documented effect,  ` nargs='+'` should be used in place of `action='append'.

1. Which is the desired behaviour? If it is as documented I'm happy to make the change. (Pretty sure this is what's wanted).

2. What is this mechanism used for? [I can see this for metric e.g. text + rodata....]

Roger-Shepherd, 2021-03-20:
The functionality as described is implemented in a fix to bench+size.py. Once accepted this will close this issue.

paoloR01, 2021-03-22:
I tested your patch and it works nicely. Maybe the documentation should clarify that all the sections listed with `--text`, `--data` etc... are then included in the respective `--metric` category `text`, `data` etc **if** the category is specified in `--metric` (unless it is the default only: `text`). I can add that to the doc if it sounds reasonable.

Roger-Shepherd, 2021-03-22:
I think it would be a good idea to make that change to the documentation.

PaoloS02, 2021-03-22:
I'll push a pull request as soon as I get the chance (btw I hadn't realized I was logged in with my other account paoloR01).

closed 2021-04-09 by Roger-Shepherd