embench/embench-iot #122: Linking the GCC and maths libraries needs a differnt order of arguments to GCC then generated by the buld_all.py script

tariqkurd-repo, 2021-03-24:
I'd like to build embench for RISCV with the gcc and maths libraries included.

here's the command line:

./build_all.py --cc /data/toolchains/riscv/riscv32-embecosm-ubuntu1804-gcc10.2.0-r4/bin/riscv32-unknown-elf-gcc --clean --verbose --arch riscv32 --chip size-test-gcc-rv32imc --cflags ' -Os -msave-restore -frecord-gcc-switches' --ldflags ' -Os -msave-restore'

here's the chip.cfg:

cflags = [
    '-march=rv32imc', '-c', '-Os', '-fdata-sections', '-ffunction-sections'
]
ldflags = [
    '-march=rv32imc', '-Os', '-Wl,-gc-sections', '-nostartfiles', '-lgcc', '-lm'
]
dummy_libs  = ['crt0']

it tries to link "cubic" with

/data/toolchains/riscv/riscv32-embecosm-ubuntu1804-gcc10.2.0-r4/bin/riscv32-unknown-elf-gcc -march=rv32imc -Os -Wl,-gc-sections -nostartfiles -lgcc *-lm* -Os -msave-restore -o cubic basicmath_small.o libcubic.o /data/benchmarks/embench/embench-iot/bd/config/riscv32/chips/size-test-gcc-rv32imc/chipsupport.o /data/benchmarks/embench/embench-iot/bd/config/riscv32/boards/generic/boardsupport.o /data/benchmarks/embench/embench-iot/bd/support/main.o /data/benchmarks/embench/embench-iot/bd/support/beebsc.o /data/benchmarks/embench/embench-iot/bd/support/dummy-crt0.o

which gives link errors 

/data/toolchains/riscv/riscv32-embecosm-ubuntu1804-gcc10.2.0-r4/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: libcubic.o: in function `SolveCubic':
libcubic.c:(.text.SolveCubic+0x33e): undefined reference to `sqrt'
/data/toolchains/riscv/riscv32-embecosm-ubuntu1804-gcc10.2.0-r4/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: libcubic.c:(.text.SolveCubic+0x364): undefined reference to `acos'

etc. 

but if I move the *-lm*  to the end (or duplicate it there) then it links

is there anyway to control the position of the arguments?

jeremybennett, 2021-03-24:
There is an option for putting things like -lm. IIRC it's `--user-libs`.  This is precisely because of the problem you have encountered with `--ldflags`.

HTH

tariqkurd-repo, 2021-03-25:
Thanks Jeremy, that's the answer

closed 2021-03-25 by tariqkurd-repo