embench/embench-iot #126: Dummy library uses "unsigned int" rather than "size_t"

Roger-Shepherd, 2021-04-04:
Trying to build the benchmarks for size on a Mac and thus using dummy-libc.c the compiler complains about (e.g.)

    int
    memcmp (const void *s1 __attribute__ ((unused)),
        const void *s2 __attribute__ ((unused)),
        unsigned int n __attribute__ ((unused)))
        {
            return 0;
        }
with the message

    /Users/Roger/Programming/embench-iot/support/dummy-libc.c:44:1: note: 'memcpy' is a builtin with type 'void *(void *, const void *, unsigned long)'

I think the problem is that the sourcing is using `unsigned int` rather than `size_t`.  I believe that _at least_ x86-64 and aarch64 (arm64) Linux, OS X and iOS all have size_t ultimately defined as unsigned long.

Roger-Shepherd, 2021-04-04:
I've made this suggested change and it works for ARM M1 and X86 compilation on Mac OS. It needs testing on other systems.

PaoloS02, 2021-04-09:
Solved with PR #132

closed 2021-04-09 by PaoloS02