embench/embench-iot #128: Problems with documentation makefile/scripts

Roger-Shepherd, 2021-04-05:
Preparing to update the `doc/README.doc` to reflect the added support for macho I have some problems with the scripts/Makefiles in `doc`.

1. Regarding spelling checking, the command line for `hunspell` is `hunspell -p custom.wordlist -l -d en_US`. I think the standard dictionary is `en-US`. With that fix
2. The spelling checker throws out a lot of words. Should I add these to the `custom.wordlist`?
3. `make too` doesn't work for me. It produces a new version of README.md with the all the text prior to 
`` removed but no new toc. I don't know how to debug this.

Roger-Shepherd, 2021-04-06:
The problem with `make toc` is that `genetic.sh` assume gnu sed and uses some gnu specific extensions.

1. `\?` which can be replaced by `\{0,1\}
2. `0,/regexp/` which can be replaced by `0,/regexp/` _unless_  `regexp` matches the very first line of input.

My inclination is to use the (e.g. Apple) sed compatible replacements rather than require gnu sed.

PaoloS02, 2021-04-09:
Solved with PR #132

closed 2021-04-09 by PaoloS02