embench/embench-iot #139: Issue Building Embench --collect2: error: ld returned 1 exit status

yanz08, 2021-07-19:
When I compiled the benchmarks using the following command:

**python3 ./build_all.py --arch riscv32 --chip generic --board ri5cyverilator –cc riscv32-unknown-elf-gcc**  

I got the error blow: 
pport to binary directory /home/embench-iot/bd/support failed
Warning: Compilation of main.c from source directory /home/embench-iot/support to binary directory /home/embench-iot/bd/support failed
Warning: Compilation of chipsupport.c from source directory /home/embench-iot/config/riscv32/chips/generic to binary directory /home/embench-iot/bd/config/riscv32/chips/generic failed
...

Part of log file:
Supplied arguments
==================
--builddir            : bd
--logdir              : logs
--arch                : riscv32
--chip                : generic
--board               : ri5cyverilator
--cc                  : riscv32-unknown-elf-gcc
--ld                  : None
--cflags              : None
--ldflags             : None
--env                 : None
--cc-define1-pattern  : None
--cc-define2-pattern  : None
--cc-incdir-pattern   : None
--cc-input-pattern    : None
--cc-output-pattern   : None
--ld-input-pattern    : None
--ld-output-pattern   : None
--user-libs           : None
--dummy-libs          : None
--cpu-mhz             : None
--warmup-heat         : None
--verbose             : False
--clean               : False
--timeout             : 5

Global parameters
=================
rootdir              : /home/embench-iot
verbose              : False
bd                   : /home/embench-iot/bd
configdir            : /home/embench-iot/config
bd_configdir         : /home/embench-iot/bd/config
archdir              : /home/embench-iot/config/riscv32
bd_archdir           : /home/embench-iot/bd/config/riscv32
chipdir              : /home/embench-iot/config/riscv32/chips/generic
bd_chipdir           : /home/embench-iot/bd/config/riscv32/chips/generic
boarddir             : /home/embench-iot/config/riscv32/boards/ri5cyverilator
bd_boarddir          : /home/embench-iot/bd/config/riscv32/boards/ri5cyverilator
env                  : {}
benchdir             : /home/embench-iot/src
bd_benchdir          : /home/embench-iot/bd/src
supportdir           : /home/embench-iot/support
bd_supportdir        : /home/embench-iot/bd/support
cflags               : ['-I/home/embench-iot/support', '-I/home/embench-iot/config/riscv32/boards/ri5cyverilator', '-I/home/embench-iot/config/riscv32/chips/generic', '-I/home/embench-iot/config/riscv32', '-DCPU_MHZ=1', '-DWARMUP_HEAT=1']
ldflags              : []
cc                   : riscv32-unknown-elf-gcc
cc_define1_pattern   : -D{0}
cc_define2_pattern   : -D{0}={1}
cc_incdir_pattern    : -I{0}
cc_input_pattern     : {0}
cc_output_pattern    : -o {0}
ld_input_pattern     : {0}
ld_output_pattern    : -o {0}
user_libs            : {}
dummy_libs           : {}
cpu_mhz              : 1
warmup_heat          : 1
timeout              : 5
ld                   : riscv32-unknown-elf-gcc

General log
===========
Warning: Compilation of beebsc.c from source directory /home/embench-iot/support to binary directory /home/embench-iot/bd/support failed
Command was:
riscv32-unknown-elf-gcc -I/home/embench-iot/support -I/home/embench-iot/config/riscv32/boards/ri5cyverilator -I/home/embench-iot/config/riscv32/chips/generic -I/home/embench-iot/config/riscv32 -DCPU_MHZ=1 -DWARMUP_HEAT=1 -o beebsc.o /home/embench-iot/support/beebsc.c

/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/lib/crt0.o: in function `.L0 ':
(.text+0x32): undefined reference to `main'
**collect2: error: ld returned 1 exit status**

Warning: Compilation of main.c from source directory /home/embench-iot/support to binary directory /home/embench-iot/bd/support failed
Command was:
riscv32-unknown-elf-gcc -I/home/embench-iot/support -I/home/embench-iot/config/riscv32/boards/ri5cyverilator -I/home/embench-iot/config/riscv32/chips/generic -I/home/embench-iot/config/riscv32 -DCPU_MHZ=1 -DWARMUP_HEAT=1 -o main.o /home/embench-iot/support/main.c

/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccgBVD65.o: in function `main':
main.c:(.text+0x10): undefined reference to `initialise_board'
/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: main.c:(.text+0x18): undefined reference to `initialise_benchmark'
/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: main.c:(.text+0x22): undefined reference to `warm_caches'
/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: main.c:(.text+0x2a): undefined reference to `start_trigger'
/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: main.c:(.text+0x32): undefined reference to `benchmark'
/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: main.c:(.text+0x40): undefined reference to `stop_trigger'
/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: main.c:(.text+0x4e): undefined reference to `verify_benchmark'
**collect2: error: ld returned 1 exit status**
...

George solved the problem by updating the version of gcc. https://github.com/embench/embench-iot/issues/97#issuecomment-705154607 I cannot solve the problem in the same way.

Do you have any idea to solve this linker error?

Abraham-Castaneda, 2021-07-21:
I'm having a very similar issue, only difference is that I am using riscv64-unknown-elf-gcc and am using a different set of compiler flags (other than that our supplied arguments are the same). I hope someone sheds some light on this!

yanz08, 2021-07-21:
When I use 
python3 ./build_all.py --arch riscv32 --chip generic --board ri5cyverilator --cc riscv32-unknown-elf-gcc **--cflags=-c** 

I got:
aha-mont64
crc32
Warning: Link of benchmark "cubic" failed
edn
huffbench
matmult-int
minver
Warning: Link of benchmark "nbody" failed
nettle-aes
nettle-sha256
nsichneu
picojpeg
qrduino
sglib-combined
slre
Warning: Link of benchmark "st" failed
statemate
ud
Warning: Link of benchmark "wikisort" failed

part of log file:
Compilation of benchmark "nbody" successful
Warning: Link of benchmark "nbody" failed

/opt/riscv-gnu-toolchain-dist-rv32imac-ilp32/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: nbody.o: in function `.L9':
**nbody.c:(.text+0x2fc): undefined reference to `sqrt'
collect2: error: ld returned 1 exit status**

In directory "/home/embench-iot/bd/src/nbody"
Command was:
riscv32-unknown-elf-gcc -o nbody nbody.o /home/embench-iot/bd/config/riscv32/chips/generic/chipsupport.o /home/embench-iot/bd/config/riscv32/boards/ri5cyverilator/boardsupport.o /home/embench-iot/bd/support/main.o /home/embench-iot/bd/support/beebsc.o
Compilation of benchmark "nettle-aes" successful

sobuch, 2021-07-22:
Hi, I think the answer is already in the thread you linked here previously - you need to also add `--user-libs="-lm"` to compile with libm which contains e.g. the function sqrt. Alternatively, you can use the dummy libraries (that would be something like `--dummy-libs="crt0 libc libgcc libm"`), but in this case only benchmarking code size would make sense.

Abraham-Castaneda, 2021-07-23:
Like sobuch said, libs="-lm" should fix the issue, it did it for me.

yanz08, 2021-07-26:
Yes, thanks you sobuch and Abraham.

closed 2021-07-26 by yanz08