embench/embench-iot #143: Reproduced baseline program size is larger than expected

bzbplumb, 2021-09-21:
I am trying to reproduce the baseline code size values. Would it be possible to get more details on how to build the code size baseline?

I am getting values larger than the baseline - see below - using the gcc compiler version 9 and 10. I have included the build and size log file.

Taking the aha-mont64 module as an example. The module code size is small ~888bytes but the total size is bigger than expected 2.38kb.

Any help would be much appreciated

[build-2021-09-17-113501.log](https://github.com/embench/embench-iot/files/7205409/build-2021-09-17-113501.log)
[size-2021-09-17-113954.log](https://github.com/embench/embench-iot/files/7205411/size-2021-09-17-113954.log)
[size-2021-09-17-113958.log](https://github.com/embench/embench-iot/files/7205412/size-2021-09-17-113958.log)
d

jeremybennett, 2021-09-22:
Hi @bzbplumb The results should be identical for identical set ups. The precise versions of compilers etc matter for this, and are specified in the the [iot-results repository](https://github.com/embench/embench-iot-results) details for each target. For example for the [Arm Cortex M4 when configured for speed](https://github.com/embench/embench-iot-results/blob/master/details/cortexm4-armv7m-gcc-9.2-o2.mediawiki).

I've assigned to @PaoloS02 who did much of the original work on ensuring reproducibility.

bzbplumb, 2021-09-30:
Hi Jeremy,

Thanks. I have included the version output of the 9 & 10 gcc compilers.

[GNU Tools for Arm Embedded Processors 9-2019-q4-major.txt](https://github.com/embench/embench-iot/files/7261769/GNU.Tools.for.Arm.Embedded.Processors.9-2019-q4-major.txt)
[GNU Arm Embedded Toolchain 10.3-2021.07.txt](https://github.com/embench/embench-iot/files/7261770/GNU.Arm.Embedded.Toolchain.10.3-2021.07.txt)

I also tried the build_all command line from the run_all.py script with the same result.
./build_all.py --builddir build --logdir logs --clean --verbose --arch=arm --chip=cortex-m4 --board=generic --cc=arm-none-eabi-gcc --cflags="-c -mcpu=cortex-m4 -mthumb -mfloat-abi=soft -Os" --ldflags="-nostartfiles -nostdlib" --dummy-libs="crt0 libc libgcc libm"

bzbplumb, 2021-10-10:
Hi Jeremy and Paolo,

I found the cause of my woes - it was all user error.  The replicated baselines values I am getting are close to the actual values. Geometric Mean=0.95 for both version 9 & 10 of the ARM gcc compiler

Command line: ./build_all.py --builddir build --logdir logs --clean --verbose --arch=arm --chip=cortex-m4 --board=generic --cc=arm-none-eabi-gcc --cflags="-c -mcpu=cortex-m4 -mthumb -mfloat-abi=soft -Os -ffunction-sections -fdata-sections" --ldflags="-nostartfiles -nostdlib -march=armv7 -Wl,-Map=arm-m4.map -Wl,--gc-sections" --dummy-libs="crt0 libc libgcc libm"

[baseline_results.csv](https://github.com/embench/embench-iot/files/7317699/baseline_results.csv)

Regards
Bill

closed 2021-10-10 by bzbplumb