embench/embench-iot #161: In the st benchmark some toolchains hoisting/removing the computation from the benchmarking loop

MarkHillHuawei, 2022-05-16:
Some of the result variables in embench-iot/st are auto-declared (MeanA, MeanB, VarA, VarB,StddevA, StddevB) and at risk of the computation being removed via DCA/loop hoisting. For example, the inner loop include 3 fp divides per iteration but when armclang/6.14.1 compiles at O3 only one fp divide  is performed in the whole benchmark run.

Re-declaring these as globals  and calling the main loop body by function pointer seems to prevent this over-optimisation of the benchmark.

Roger-Shepherd, 2022-05-16:
I understand why Re-declaring these as globals and calling the main loop body by function pointer might prevent the over optimisation, but what other effects do the changes have? I'd expect performance to go down and code size to go up. Have you measured this?

Roger-Shepherd, 2022-05-16:
Sorry, wasn't clear, I meant - apart from the removal of the "over optimisation" what was the input.