embench/embench-iot #171: nsichneu use of volatile keyword

axiomlegend, 2022-09-13:
Hello all;
These variables, marked volatile, inhibit compiler optimization of the code. Comments mention that the variables are so marked to prevent the compiler from simplifying the computation (true).   

>volatile int P1_is_marked = 3;
>volatile long P1_marking_member_0[3];
>volatile int P2_is_marked = 5;
>volatile long P2_marking_member_0[5];
>volatile int P3_is_marked = 0;
>volatile long P3_marking_member_0[6];

There is a better way to ensure the computation is performed, with some optimization. Instead of using volatile,  initialize the P1/P2/P3 is_marked variables from  an array at the beginning of the loop and write their values back to the array at the end of the loop.  Compilers can then optimize the code.

I believe this change will enable this benchmark to more accurately reflect generated code and performance of a petri-net.

jeremybennett, 2022-09-14:
Hi @axiomlegend Thanks for the suggestion

One of the things we try to do with Embench is to leave the programs as we find them as far as is possible. The only changes we make are to ensure they work on a range of architectures from 8-bit to 64-bit and occasionally to fix bugs that mean they won't work. TBH the biggest change is usually fixing assumptions about the size of value that can fit in an `int`.  On some small architectures, `int` is only 16-bits long.

The programs are far from perfect, but that is deliberate. The last thing we want is a set of programs rewritten to what we believe the perfect solution would be, since that doesn't reflect the real world.

jeremybennett, 2022-09-19:
Marking as closed.

closed 2022-09-19 by jeremybennett